ExpressLRS/ExpressLRS-Hardware #5: Gerber file not working for pp_rx

dkuchay, 2021-05-26:
Attempted to order the PP_RX from JLCPCB but ran in to a problem that instructions did not cover.

Gerber file, bom file and cpl file loaded to jlc however the item of PAD_1MM kicked out saying it didnt make sense to their inventory system. I chat in to their help system and asked someone to look the files over. They, "Mitchell" confirmed that Pad_1mm is not an item they can find either. Mitchell also asked to make the bridges that connect the 20mm frame to the 10mm board at least 4mm otherwise it may not manufacture cleanly. I searched discord but only found mention of the item during what looked like design time verbosity. Also I ordered all the items in the bom for rx_nano but upon sorting out it looks like I bought a number of parts that are not used. Inductors to name one. Are these files current? I wasted a lot of time and money on parts I wont use.

To reproduce either load the gerber file, cpl and bom files to jlcpcb and attempt to mark for jlc assembly and press next and or download the bom for rx_nano and compare to the image that is left for hints on layout. The items pictured do not match the bom.

maybenikhil, 2022-01-07:
Pad_1mm is just a pad that's on the PCB and can be removed from the BOM

closed 2022-01-07 by maybenikhil

dkuchay, 2022-01-07:
Dude, it’s like 8 months later. Never mind.

maybenikhil, 2022-01-07:
I'm aware, was just doing some winter cleaning. No one actively checks issues on this repo. Sorry about that!

dkuchay, 2022-01-07:
No worries, I’ve worked support. Close close close