ExpressLRS/ExpressLRS-Hardware #15: DIY RX_20x20 PCB 2400MHz different capacitor values?

ucflumm, 2021-08-08:
In the readme it says C2 3528 47uF Tantalum Capacitor.

But when looking at the schematic its a 100uF capacitor?

ucflumm, 2021-09-04:
PnP, partslist and schematic have all different values to the readme for the v2.0 2020rx

JBKingdon, 2021-09-11:
The capacitors aren't an issue. Use any pair of capacitors between 10 and 100uF.

However, there are lots of other problems in the readme relating to the resistors. The 10k pullups were removed from this revision, but we have both a pullup and an in-line current limiting resistor on the esp TX line, and I don't think the equal values shown in the schematic will work. Wasn't this fixed before? I think it would work if R5 >> R3, e.g. R5 = 470, R3 = 33 ohms or something, but I haven't tested. For completeness, R4 is the current limiting resistor for the LED.