ExpressLRS/ExpressLRS-Hardware #24: DIY RX ESP8266

royyandzakiy, 2021-11-28:
Hi! I've been wondering, can I flash the DIY_900_RX_ESP8285_SX127x on an ESP8266 Lolin V1 board connected to an RFM95?

I currently don't have an ESP8285 so I was wondering if this will work out. If I'm not mistaken, they actually are the same boards, the only difference is that the ESP8285 has a 1 MB flash chip right within the chip

Really appreciate your time reading! Thanks!

kimman1, 2022-01-05:
In my country. I've seen a boy who redesign PCB of RFM95 with ESP8255 to ESP8266 and it's working so as think It will work with no issue.

royyandzakiy, 2022-01-06:
> In my country. I've seen a boy who redesign PCB of RFM95 with ESP8255 to ESP8266 and it's working so as think It will work with no issue.

I see, if you perhaps any information about the PCB please let me know. Thanks for the reply!

OpenUAS, 2025-01-22:
@royyandzakiy asked this at Nov 28, 2021 and also pondered "I actually don't know if this is the right forum to ask", so true indeed. Maybe  maintainers should take the liberty to close this issue?