ExpressLRS/ExpressLRS-Hardware #30: Connecting pins

Multisoft1978, 2022-02-01:
Hello
I don't understand how it's connected.
Can I connect it to Taranis Q X7?
TX - SBUS/PPM ?
3.3V
GND
RX -> S-PORT?

kkbin505, 2022-02-10:
This is an receiver (RX) not TX, you can not connect it to transmitter. You can only connect it to a FC.

mariusdp, 2022-10-07:
If you mean diy TX to QX7, you connect only 3 wires. V+, GND and SPORT.

Multisoft1978, 2022-10-07:
Yes, I already figured it out.  Now me another question torments, why does a 500mW module in a DIY project work at a maximum of 250mW?  :)

mariusdp, 2022-10-31:
has to do with stability/performance. That is the max power output ELRS supports for now...

OpenUAS, 2025-01-22:
The assumption "why does a 500mW module in a DIY project work at a maximum of 250mW?" is incorrect, having a 1000mW module in one of our DIY projects works at a maximum of 1004mW. @Multisoft1978 Having answered the second question, as asked in this in non related base issue #30, which is not a ELRS Hardware issue per se, IMHO now is the time to close this issue.