ExpressLRS/ExpressLRS-Hardware #42: Reference Design

bennymeg, 2022-06-28:
I would like to design my own transmitter and receiver.
Do you have a reference design or minimal schematic that I can base on?

kkbin505, 2022-10-08:
Do you want to build a module or a full transmitter?

bennymeg, 2022-10-08:
I would like to build a transmitter, and I would like to use supported MCU & transmitter ic combination.

kkbin505, 2022-10-09:
> I would like to build a transmitter, and I would like to use supported MCU & transmitter ic combination.

Check out my repo
https://github.com/kkbin505/Arduino-Transmitter-for-ELRS/tree/VSCODE

It use arduino nano, and is the simplest desgin I can implement

bennymeg, 2022-10-09:
Thank you, but what I meant is that I want to build the transmitter (like you built a DIY one) that I can download an unmodified firmware from the ELRS configurator. So, what I am actually what to know is where can I find a description of the firmware compatibility.