ExpressLRS/ExpressLRS-Hardware #46: DIY TX documentation issue

gusarov, 2022-11-06:
On the page
https://github.com/ExpressLRS/ExpressLRS-Hardware/tree/master/PCB/2400MHz/TX_SX1280

after reading everything, I noticed that there is 2 PCB design, and no hint about which one to choose and why, so I just took a first one (G-NiceRF_LoRa1280). And ordered all the parts, including E28. While soldering it turned out that E28 module can't fit onto LoRa1280 (I just never came across that before, so had no idea that this part of file name have something to do with different module at all). This is quite sad and looks like a serious issue with the doc. Let's make it more clear, both in the list of variants of PCB and in the list of parts to order. There is nothing about LoRa1280 on this page at all, I just learned it hard way.

sl1pkn07, 2023-06-28:
seems https://es.aliexpress.com/item/1005004877348173.html

gusarov, 2023-07-08:
Consider to not post localized versions of a link. Due to Ali behavior, whoever clicks on that will switch their cookies and will see all pages in Spanish till they manually change the profile settings...

sl1pkn07, 2023-07-08:
done